scarburato/t150_driver #2: T300RS module in the works

berarma, 2020-08-06:
Hi.

I'm happy to see you're working on getting Thrustmaster wheels fully supported on Linux. That would be great. Thanks!

I know some who's writing a T300RS module. Here's his project: [hid-tmff2](/Kimplul/hid-tmff2).

I guess your projects could mutually benefit from sharing information since both wheels may have a lot of similarities.

I don't own a TM wheel myself but I've written an improved module for Logitech wheels: [new-lg4ff](/berarma/new-lg4ff). I've also written some tools: [Oversteer](/berarma/oversteer), [ffbtools](/berarma/ffbtools). Maybe these can be valuable to you.

Cheers.

scarburato, 2020-08-07:
Thanks! I've already used your ffbwrapper to log Assetto Corsa's effects, I've found it very useful.

It seems that the T300RS protocol to upload ffb effects is quite different from the T150; but it can be quite useful if I decide to re-write my driver as an hid_driver like I should have done from the start!

Regarding Oversteer: to add support for my wheel do I have to rename the sysfs attributes to match what Oversteer expects or vice versa?

berarma, 2020-08-07:
> It seems that the T300RS protocol to upload ffb effects is quite different from the T150; but it can be quite useful if I decide to re-write my driver as an hid_driver like I should have done from the start!

Having a unified module driver like the Logitech module would be best as long as possible.

> Regarding Oversteer: to add support for my wheel do I have to rename the sysfs attributes to match what Oversteer expects or vice versa?

Better to match the existing sysfs attributes, it'll be automatically recognized by Oversteer and any other existing tool like pyLinuxWheel. I'll just need to add the wheel vendor/model id.

albfan, 2023-03-19:
Hi, sysfs attributes should be fixed by d8ca058fa0a301aeb688ebeadbf6fca281ed3f9f.

About oversteer and different vendors, I think is unfair to ask driver to remap things (although there's a standard anywhere saying what axis or buttons should report what) and oversteer needs to deal with that using a mapping based on vendor/model

I'm working on that in https://github.com/berarma/oversteer/issues/131

To my understanding, hid_t150 state of the art is in sync with oversteer, so this can be closed.

Still some functions don't do exactly what they should, but I suggest to drive that with new specific issues

berarma, 2023-03-19:
> Hi, sysfs attributes should be fixed by [d8ca058](https://github.com/scarburato/t150_driver/commit/d8ca058fa0a301aeb688ebeadbf6fca281ed3f9f).
> 
> About oversteer and different vendors, I think is unfair to ask driver to remap things (although there's a standard anywhere saying what axis or buttons should report what) and oversteer needs to deal with that using a mapping based on vendor/model
> 
> I'm working on that in [berarma/oversteer#131](https://github.com/berarma/oversteer/issues/131)
> 
> To my understanding, hid_t150 state of the art is in sync with oversteer, so this can be closed.
> 
> Still some functions don't do exactly what they should, but I suggest to drive that with new specific issues

This issue isn't about the mapping of buttons/axes, sysfs or T150 features. And I think you misunderstood my last comment where I wasn't talking about unifying the mapping of axes/buttons but the merging of projects to have only one module for Thrustmaster wheels if possible.

The point of this issue was talking about the feasibility of merging this module with the T300 and have one module for many Thrustmaster wheels instead of one module per wheel model. Or at least try to reuse as much code/knowledge as possible between projects.

albfan, 2023-03-19:
Then we are on same path. After make this driver work I want to contact kernel maintainer to integrate this so it comes out of the box.

kernel provides hid_tmff, but looks there was no try to unify hid_tmff2 with it.

https://github.com/Kimplul/hid-tmff2/issues/35#issuecomment-1022434773

we can contact hid_tmff2 to see if that can be done (I can work on that) and then see if hid_tmff2 should be upstreamed or integrated in hid_tmff. I'm unsure about the best way to maintain all this, maybe is to use 3 drivers

albfan, 2023-11-13:
No progress, so let's close by now

closed 2023-11-13 by albfan