scarburato/t150_driver #10: Can't install on Manjaro

Reymai, 2020-12-12:
After launching ./install.sh, I have this console log:
```
==== REMOVING OLD VERSIONS ====
./install.sh: line 13: dkms: command not found
==== CONFIG DKMS ====
mkdir: cannot create directory ‘/usr/src/t150-0.7b’: File exists
mkdir: cannot create directory ‘/usr/src/t150-0.7b/build’: File exists
mkdir: cannot create directory ‘/usr/src/t150-0.7b/hid-tminit’: File exists
fatal: destination path '/usr/src/t150-0.7b/hid-tminit' already exists and is not an empty directory.
==== DKMS ====
./install.sh: line 32: dkms: command not found
./install.sh: line 33: dkms: command not found
./install.sh: line 34: dkms: command not found
==== SET UP LOAD AT BOOT ====
==== INSTALLING UDEV RULES ====
'./files/etc/udev/rules.d/10-t150.rules' -> '/etc/udev/rules.d/10-t150.rules'
==== LOADING NEW MODULES ====
modprobe: FATAL: Module hid-tminit not found in directory /lib/modules/5.9.11-3-MANJARO
modprobe: FATAL: Module hid-t150 not found in directory /lib/modules/5.9.11-3-MANJARO
```

scarburato, 2020-12-13:
>./install.sh: line 13: dkms: command not found

You have to install `dkms` first

Reymai, 2020-12-13:
Thank you very much!

closed 2020-12-13 by Reymai