scarburato/t150_driver #12: How to test

toft864, 2021-02-05:
Hi, 

Did the install, dmesg gives this output - I'm trying to use the wheel in rfactor 2 thru steam, running manjaro - but can't get any response from the wheel or pedals. 

Any way to actually test that it works ?

[87440.388999] usb 1-2: new full-speed USB device number 13 using xhci_hcd
[87440.530647] usb 1-2: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[87440.530653] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[87440.530656] usb 1-2: Product: Thrustmaster FFB Wheel
[87440.530659] usb 1-2: Manufacturer: Thrustmaster
[87440.534048] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:044F:B65D.000F/input/input46
[87440.534439] hid-tminit 0003:044F:B65D.000F: input,hidraw5: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:00:14.0-2/input0
[87440.553260] hid-tminit 0003:044F:B65D.000F: Wheel with model id 0x306 is a Thrustmaster T150RS
[87440.556380] hid-tminit 0003:044F:B65D.000F: Success?! The wheel should have been initialized!
[87440.556485] usb 1-2: USB disconnect, device number 13
[87440.859011] usb 1-2: new full-speed USB device number 14 using xhci_hcd
[87441.000453] usb 1-2: New USB device found, idVendor=044f, idProduct=b677, bcdDevice= 1.00
[87441.000460] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[87441.000463] usb 1-2: Product: Thrustmaster T150RS
[87441.000466] usb 1-2: Manufacturer: Thrustmaster
[87441.003927] hid-t150 0003:044F:B677.0010: hidraw5: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T150RS] on usb-0000:00:14.0-2/input0
[87441.004027] input: Thrustmaster T150 steering wheel as /devices/pci0000:00/0000:00:14.0/usb1/1-2/input/input47
[87441.017127] hid-t150 0003:044F:B677.0010: Setup completed! Firmware version is 21

scarburato, 2021-02-06:
To test the input run ```evtest```, also take note of the device path (ex. ```/dev/input/event19``` ).
To test the forcefeedback run ```fftest [device path]``` and ```ffmvforce [device path]```, on Ubuntu those two programs are found in the ```joystick``` packet.

> I'm trying to use the wheel in rfactor 2 thru steam

Anyway a Reddit user reported having problem with Steam installed as a flatpack, maybe is your case? ([link](https://old.reddit.com/r/linux_gaming/comments/i4xrk2/im_writing_a_kernel_driver_for_the_thrustmaster/g5plcdt/))

toft864, 2021-02-07:
evtest, fftest and ffmvforce all works fine. Steam is installed thru the manjaro package - but still no luck with rfactor.

berarma, 2021-02-07:
Switch off all other controllers. It might be grabbing them instead of the wheel. You can also redefine the controls in the game. I've had that problem even with a graphics tablet.

toft864, 2021-02-07:
Is it ok that it's detected as both /dev/input/event19 and /dev/input/js0 ?
Should the controller settings section in Steam detect it ?

dpisklov, 2021-02-09:
@toft864 if you're using Manjaro with KDE like myself, there's a section in System Settings -/ Input Devices -/ Game Controller, the wheel is there and showing all the axes as well as the buttons.
For me it worked like a charm in Manjaro, I use it with ATS - great work @scarburato

albfan, 2023-03-19:
> evtest, fftest and ffmvforce all works fine

This is then a steam problem, not a driver problem. I suggest to close this

closed 2023-03-20 by scarburato