scarburato/t150_driver #13: Clutch axis swapped with wheel axis

86kotek, 2021-03-12:
With just `hid-tminit` the wheel works alright but no FFB, when i want to use the driver tho the axis of clutch and wheel is swapped.
I dont know much about coding but when i was fiddling with the files i swapped those axis but now both are out of range, wheel reads only values when turning left

86kotek, 2021-03-12:
I thought i provide more info on my issue:
`evtest` runs ok, shows that clutch is swapped with gas, which is no problem, maybe Debian game controller configuration reads something wrong because wheel is on Y axis and gas on X axis
`ffmvforce` works as expected
In `fftest` those 3 have no effect on the wheel:
`Uploading effect #1 (Constant) ... OK (id 1)
Uploading effect #2 (Spring) ... OK (id 2)
Uploading effect #3 (Damper) ... OK (id 3)`
test 0, 1 and 5 works

Debian Game Controller sees my wheel as:
`Thrustmaster T150 steering wheel (/dev/input/js0)` although as said before wheel is on Y axis and gas on X axis
Steam sees it as Generic Gamepad: Generic Gamepad
Speaking of Steam i didnt installed it from flatpack as found in another issue.

`lsmod | grep hid-t150
lsmod | grep hid-tminit
does not show anything but there is ` hid_t150` and `hid_tminit` when
`lsmod | grep hid`
`hid                   147456  6 hid_t150,hid_tminit,usbhid,hid_generic,hid_logitech_dj,hid_logitech_hidpp`

`dkms status` shows 

`t150, 0.7c, 5.10.0-4-amd64, x86_64: installed`

`udevadm info /sys/class/input/event20/device` :

`P: /devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13.2/input/input49
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13.2/input/input49
E: PRODUCT=3/44f/b677/100
E: NAME="Thrustmaster T150 steering wheel"
E: PHYS="usb-0000:00:14.0-13.2/input0"
E: PROP=0
E: EV=20000b
E: KEY=30000 fdb000100000000 0 0 0 0
E: ABS=30704
E: FF=11c2f0000 0
E: MODALIAS=input:b0003v044FpB677e0100-e0,1,3,15,k120,130,131,133,134,136,137,138,139,13A,13B,150,151,ra2,8,9,A,10,11,mlsf50,51,52,53,55,5A,5B,5C,60,w
E: SUBSYSTEM=input
E: USEC_INITIALIZED=4413877
E: ID_INPUT_JOYSTICK=1
E: ID_INPUT=1
E: ID_SERIAL=noserial
E: ID_PATH=pci-0000:00:14.0-usb-0:13.2
E: ID_PATH_TAG=pci-0000_00_14_0-usb-0_13_2
E: ID_FOR_SEAT=input-pci-0000_00_14_0-usb-0_13_2
E: TAGS=:seat:
E: CURRENT_TAGS=:seat:`

Also `/etc/udev/rules.d/10-t150.rules` exists and consists of :
`  GNU nano 5.4                                                                                           /etc/udev/rules.d/10-t150.rules                                                                                                    
# make the t150 appear like a joystick
SUBSYSTEM=="input", ATTRS{name}=="Thrustmaster T150 steering wheel", MODE="0666", ENV{ID_INPUT_JOYSTICK}="1"

# Set up 
#SUBSYSTEM=="input", ATTRS{name}=="Thrustmaster T150 steering wheel", MODE="0666", ENV{ID_MODEL_ID}="b677"
#SUBSYSTEM=="input", ATTRS{name}=="Thrustmaster T150 steering wheel", MODE="0666", ENV{ID_VENDOR_ID}="044f"`

scarburato, 2021-03-13:
Have you tried to disable Steam Input (game proprieties -> controller) ? Also which version of the firmware is the wheel running?

>In fftest those 3 have no effect on the wheel:
Uploading effect #1 (Constant) ... OK (id 1) Uploading effect #2 (Spring) ... OK (id 2) Uploading effect #3 (Damper) ... OK (id 3)

Are you sure? They don't play any noticeable effect immediately but the spring effect should make the wheel turn back to center when turned. See more [here](https://support.thrustmaster.com/en/kb/107-en/)

86kotek, 2021-03-14:
> Have you tried to disable Steam Input (game proprieties -> controller) ? Also which version of the firmware is the wheel running?
> 
> > In fftest those 3 have no effect on the wheel:
> > Uploading effect #1 (Constant) ... OK (id 1) Uploading effect #2 (Spring) ... OK (id 2) Uploading effect #3 (Damper) ... OK (id 3)
> 
> Are you sure? They don't play any noticeable effect immediately but for example the spring effect should make the wheel turn back to center when turned. See more [here](https://support.thrustmaster.com/en/kb/107-en/)

Hi, yes i did tried disabling Stem Input but with no effect. 
I just bought the wheel and dont know how to check firmware in Linux to be honest, frankly i only switched to Linux couple weeks ago
Effect 1,2 and 3 are working actually, dont know why they didnt before but its ok now
I also noticed in KDE game controller device is listed as
'/dev/input/js0'  - i cant run any tests on it
but `evtest` lists it as
`/dev/input/event24`- i can run tests on it

86kotek, 2021-03-17:
> Have you tried to disable Steam Input (game proprieties -> controller) ? Also which version of the firmware is the wheel running?
>
>
> Are you sure? They don't play any noticeable effect immediately [...]

So i realised i was looking at the wrong numbers and my firmware was actually really old, updated the firmware from v5 to v21 with no immediate changes at all but i reinstalled the whole system and now Iam running clean Debian Bullseye and the wheel does more than before.
With previous installation ETS2 was working sort of ok, and other games wouldn't, now it doesn't work at all in ETS2 BUT it works quite alright in Assetto Corsa, which Iam really happy off :)
In Dirt Rally 2.0  as soon i start the game, the wheel pulls to the left and more importantly has a very strong rumble effect constantly, no matter the settings in game, even if the vehicle is not moving. That works with Steam input On , as xbox360 controller but with Input disabled its not being recognised at all in Dirt.
Any ideas?

closed 2021-03-23 by 86kotek