scarburato/t150_driver #14: Steam & BeamNG

sniperlucian, 2021-03-21:
Its working within Steam with Assetto Corsa. But BeamNG doesn't pick it up. It finds the xbock controller just fine - just not the wheel. Also tried all the controller settings with Steam. 

Don't know it's something that can be solved with this driver though. But maybe someone knows a solution.

scarburato, 2021-09-22:
Try the `hid` branch. It's a problem with Proton after a certain version (5.2 I think), the compatibility layer, but the `hid` branch seems to go around the Steam issue as a side effect

Friz64, 2021-09-22:
Just had a long BeamNG.drive gaming session.. without any issues! I am grateful for your efforts :)

albfan, 2023-03-19:
I think this issue can be closed