scarburato/t150_driver #16: Weird FFB behaviour

86kotek, 2021-03-23:
In Dirt Rally 2.0  as soon i start the game, the wheel pulls to the left and more importantly has a very strong rumble effect constantly, no matter the settings in game, even if the vehicle is not moving. That works with Steam input On , as xbox360 controller but with Input disabled its not being recognised at all in Dirt.
Any ideas?

icarusfactor, 2021-05-08:
With jstest CLI program all the TS150 driver options work. Gamepad - Wheel Axis and both pedal Axis. 
Would like to add this driver to the Oversteer program. 
Mine is recognized as Thustmaster T150RS in Steam
Also I have Steam Racing Games:
DirtRally1.0 - Only works with my XBOX controller. I made a 3D printed wheel to work well with it. 
DIRT4 -- The normal TS150 gamepad buttons work, but the Axis for wheel or either pedal do not work.  100% rumble while in game.
F1 2017 -- Only works with my XBOX gamepad. 

No ideas yet. Just posting my report.

isopix, 2021-05-08:
General knowledge is that wheel works only in proton or SDL2-enabled
wine builds.
Dirt 1.0 and f1 2017 got native version, not sure about Dirt 4. But
remember that it may need corrdct wheel profile, and it's possible
that it only got for G25/G27and G29, because there were only
linux-supported wheels then.
At least check if you got correct udev rules and permissions to access
wheel from your user.
Use jstest and ffbtest(FFB needs /dev/input/eventX.
You can temporarily check permissions(until reboot/wheel reconnection)
but chmod a+rwX -R /dev/input/

icarusfactor, 2021-05-08:
Thanks , going to sleep now, but thought I would do my due diligence and report my issues.
 It is like a failed config setup and since gamepad config in STEAM settings has to be disabled, yes, its like it installed for a different wheel version and does not let me modify the AXIS for DIRT4.  Will debug further later on today.

albfan, 2023-03-19:
I think all this reports should start on some steam issue tracker and reach here with some specific technical info to check what is working wrong.

In any case, this looks like "non actionable" as we don't have info to check what to fix. Suggesting to close, and reopen if new info is provided or this is test again with latest version of the driver

closed 2023-03-31 by albfan