scarburato/t150_driver #19: How can I fix the clutch axis?

culebron, 2021-03-30:
Hello everyone,
grazie mille, Dario, sei grandissimo! Lots of thanks to Dario and other contributors! You saved my day.

I got such a wheel recently and it worked well for me, except for clutch pedal. (Model T150 PRO, Live For Speed simulator detects  clutch pedal as another brake).

I have a bit of C experience with Arduinos. What can I change to try recompiling it and making clutch axis work independently?

scarburato, 2021-03-30:
The js device is created by the legacy joystick api, it is preferable to use event. You can test an event device by running `evtest` and selecting the wheel. Are in `evtest` the clutch, brake or steering reported on the same axis? 

> I see t150_input_state_packet, but if it's what the wheel sends, then IDK how to change that, except send it some state packet?

It is the content of the packet received from the wheel. I have the T150 without clutch and gear stick so I guessed where the clutch was and probably I've made the wrong guess 😅.

If the problem is real (not caused by the game) we have to find out where the clutch axis is reported in packet, to do so install Wireshark, load the the `usbmon` kernel module with `modprobe usbmon`, start capture session on `usbmon0` and start moving the clutch pedal then stop the recording. Attach the recording to this discussion. We are looking for packets starting with 07, so you can apply the filter `usb.capdata[0] == 07` and start looking for which bytes change when the clutch position is modified

culebron, 2021-03-31:
Ok, I did
* install usbmon
* sudo modprobe usbmon
* install wireshark
* sudo wireshark

Here are two capture files on `usbmon0`: one with LFS running, another without it.

https://drive.google.com/file/d/1ucttnsIxm1NY1If8pljLmS7Tx5ai0Tq6/view?usp=sharing
https://drive.google.com/file/d/1Y2GgadQqV6IxpTHKNUBc3aJEqZhuRzZz/view?usp=sharing

culebron, 2021-03-31:
[image]
[image]
[image]
[image]

culebron, 2021-03-31:
Seems to be the same as expected in `input.h`.

But it's not LFS issue, the same happens in Wine controllers setup: 
* "event" device puts brake and clutch on Z, Rz, 
* "js" device puts clutch on X/Y

Is that a problem with just axes indicators? (https://elixir.bootlin.com/linux/latest/source/include/uapi/linux/input-event-codes.h#L824)

[image]
[image]

culebron, 2021-03-31:
FIXED IT!

* changed ABS_Z to ABS_RUDDER
* make
* sudo rmmod hid-t150.ko
* sudo insmod hid-t150.ko
* dmesg -w
* unplug & plug the wheel
* everything's fine in dmesg
* new axis in LFS!

scarburato, 2021-03-31:
0k thanks! Can you make a merge request with the change?

Also, if you'd like you can make the same experiment to find the shifter buttons

culebron, 2021-03-31:
Made a pull request for the pedal. To make it appear in Wine controller tool, I set it to RZ.

culebron, 2021-03-31:
Checked the buttons, all of them work fine.

closed 2021-03-31 by scarburato