scarburato/t150_driver #21: Issues with installation [arch]

maalos, 2021-05-07:
after running `sudo sh install.sh` seems like it has installed, however, when i run `lsmod | grep hid-tminit` or `lsmod | grep hid-t150` i am not getting any output. dirt rally doesn't see my wheel, and ets2mp doesn't even start. the wheel doesn't have autocenter on so i'm guessing the drivers are here? it was working a week ago, but i messed up with my x server and drivers, so i had to do a clean install.

scarburato, 2021-05-07:
do you have dkms installed?

maalos, 2021-05-07:
`sudo pacman -S dkms`? it is installed

scarburato, 2021-05-07:
Run

```dkms status```

Can you see the t150 module ?

maalos, 2021-05-07:
yep
[image]

scarburato, 2021-05-07:
🤔

Try to attach the wheel and post the ```dmesg``` output

maalos, 2021-05-07:
`evtest` does detect my wheel. it "sees" every movement as it should - the wheel rotation, buttons and pedals
```
[  905.360772] audit: type=1130 audit(1620409689.161:136): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-tmpfiles-clean comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  905.360784] audit: type=1131 audit(1620409689.161:137): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-tmpfiles-clean comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  997.605960] usb 1-1.2: new full-speed USB device number 5 using ehci-pci
[  997.704717] usb 1-1.2: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[  997.704730] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  997.704735] usb 1-1.2: Product: Thrustmaster FFB Wheel
[  997.704738] usb 1-1.2: Manufacturer: Thrustmaster
[  997.708800] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.0/0003:044F:B65D.0008/input/input40
[  997.709162] hid-generic 0003:044F:B65D.0008: input,hidraw0: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:00:1a.0-1.2/input0
[  997.797413] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.0/0003:044F:B65D.0008/input/input41
[  997.797663] hid-tminit 0003:044F:B65D.0008: input,hidraw0: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:00:1a.0-1.2/input0
[  997.817529] hid-tminit 0003:044F:B65D.0008: Wheel with model id 0x306 is a Thrustmaster T150RS
[  997.820672] hid-tminit 0003:044F:B65D.0008: Success?! The wheel should have been initialized!
[  997.926461] usb 1-1.2: USB disconnect, device number 5
[  998.182610] usb 1-1.2: new full-speed USB device number 6 using ehci-pci
[  998.281984] usb 1-1.2: New USB device found, idVendor=044f, idProduct=b677, bcdDevice= 1.00
[  998.281997] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  998.282001] usb 1-1.2: Product: Thrustmaster T150RS
[  998.282004] usb 1-1.2: Manufacturer: Thrustmaster
[  998.283767] hid-t150 0003:044F:B677.0009: hidraw0: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T150RS] on usb-0000:00:1a.0-1.2/input0
[  998.283852] input: Thrustmaster T150 steering wheel as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/input/input42
[  998.299304] hid-t150 0003:044F:B677.0009: Setup completed! Firmware version is 11
[ 1009.739384] radeon_dp_aux_transfer_native: 242 callbacks suppressed
[ 1214.245173] usb 1-1.2: USB disconnect, device number 6
[ 1214.245357] hid-t150 0003:044F:B677.0009: T150RS Wheel removed. Bye
[ 3206.423325] audit: type=1100 audit(1620411990.219:138): pid=30118 uid=1000 auid=1000 ses=1 msg='op=PAM:authentication grantors=pam_faillock,pam_permit,pam_faillock acct="maalos" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[ 3206.425513] audit: type=1101 audit(1620411990.219:139): pid=30118 uid=1000 auid=1000 ses=1 msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="maalos" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[ 3206.426865] audit: type=1110 audit(1620411990.222:140): pid=30118 uid=1000 auid=1000 ses=1 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[ 3206.428910] audit: type=1105 audit(1620411990.222:141): pid=30118 uid=1000 auid=1000 ses=1 msg='op=PAM:session_open grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[ 3208.993737] audit: type=1106 audit(1620411992.789:142): pid=30118 uid=1000 auid=1000 ses=1 msg='op=PAM:session_close grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[ 3208.993926] audit: type=1104 audit(1620411992.789:143): pid=30118 uid=1000 auid=1000 ses=1 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[ 3799.413406] perf: interrupt took too long (2522 > 2500), lowering kernel.perf_event_max_sample_rate to 79200
[ 4336.742817] audit: type=1100 audit(1620413120.537:144): pid=32230 uid=1000 auid=1000 ses=1 msg='op=PAM:authentication grantors=pam_faillock,pam_permit,pam_faillock acct="maalos" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[ 4336.744614] audit: type=1101 audit(1620413120.537:145): pid=32230 uid=1000 auid=1000 ses=1 msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="maalos" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[ 4336.746091] audit: type=1110 audit(1620413120.540:146): pid=32230 uid=1000 auid=1000 ses=1 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[ 4336.748303] audit: type=1105 audit(1620413120.540:147): pid=32230 uid=1000 auid=1000 ses=1 msg='op=PAM:session_open grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[ 4336.759061] audit: type=1106 audit(1620413120.554:148): pid=32230 uid=1000 auid=1000 ses=1 msg='op=PAM:session_close grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[... cut ...] ss'
[ 4495.749593] audit: type=1110 audit(1620413279.543:151): pid=32441 uid=1000 auid=1000 ses=1 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[ 4495.750575] audit: type=1105 audit(1620413279.543:152): pid=32441 uid=1000 auid=1000 ses=1 msg='op=PAM:session_open grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
```

scarburato, 2021-05-07:
0k, everything is working. The problem is with Steam then. Did you install the flatpack version?

scarburato, 2021-05-07:
Last thing: run ```udevadm info  /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/input/input44```

(replace the path after ```/sys``` with the last reported in dmesg ```input: Thrustmaster T150 steering wheel as ...```)

maalos, 2021-05-07:
> 0k, everything is working. The problem is with Steam then. Did you install the flatpack version?

no, i used multilib
[image]

scarburato, 2021-05-07:
```ID_INPUT_JOYSTICK=1```

All good here, the problem is not in the driver

maalos, 2021-05-07:
you want me to deal with it on my own, huh?

scarburato, 2021-05-08:
Yes

Anyway, I remember having problems with Steam Input, try disable it (Game's proprieties -> Controller -> Disable Steam Input)

closed 2021-05-08 by scarburato