scarburato/t150_driver #24: Everything seems to be installed and working right but the wheel is not recognized in my steam games

Benjamin5678, 2021-06-05:
This [reddit thread](https://www.reddit.com/r/linux_gaming/comments/i4xrk2/im_writing_a_kernel_driver_for_the_thrustmaster/g5plcdt/) I saw in another post basically exactly describes what I have. I installed in Pop!_OS with the install.sh `lsmod | grep hid-t150` and `lsmod | grep hid-tminit` don't show anything. If I run `dkms status` I see `t150, 0.7c, 5.11.0-7614-generic, x86_64: installed`. ls mod shows it's installed also and running ev test I can see the outputs from the wheel as I move the axis. It shows as a joystick when I do `udevadm info /sys/class/input/event26/device`. What fixed this reddit users issue was installing deb version of steam which I'm already doing do I'm not sure what to do. It doesnt even register in a native game like ets2.

Lazalatin, 2021-08-05:
Hi Benjamin, 
I had a hard time finding the modules myself. Did you try `lsmod | grep hid` and looked for the appropriate module names?`
On my side there was in issue loading `hid-tminit` which I solved by adding the modules to my systemd configuration.
I am not sure if Pop!_OS is using systemd though.

scarburato, 2021-09-22:
Try the `hid` branch

adamRHargreaves, 2021-10-01:
Hi
I'm having the same issue as Benjamin.  PopOS! 21.04 here.  EVTest, FFTest all work perfectly fine, but no games (native or otherwise) pick up the wheel at all.  (have tried SuperTuxKart for native, and PCars2 / Rfactor 1 on Steam (deb version)
Have tried master and hid branch, with same result.

Please let me know if there's anything I can log which might help?  

(btw, I have the T3PA Pro 3-pedal set - all 3 pedals recognized in EVTest, so that's great!)

Benjamin5678, 2021-12-07:
How do you add modules to modules-load?

Benjamin5678, 2021-12-07:
Hey, I just came back to trying to get this to work after a long time. Is it just supposed to work like it does in windows where the game lets you bind the inputs and the wheel should register? I tried installing the hid branch, updating firmware to 35 in windows, and I'm using the automatically selected proton version for beamng. It still doesnt recognize any inputs from the controller in the game though. The evtest works with every axis and button press. I had oversteer before and it showed the device as an input but it would only recognize a few buttons and now after updating oversteer to latest and the driver to latest hid branch the drop down doesnt even show anything. Am I supposed to write anything in the /etc/udev/rules.d/10-t150.rules file? There is commented out lines in it.

adamRHargreaves, 2021-12-08:
Did you try disabling Steam Input for BeamNG?  (it's in the right-click, properties dialog for the game somewhere).  I found that no steam games could see my wheel with it enabled.
I found it was also worth trying a linux native game (SuperTuxKart for example), to see that games can see the wheel and thus prove whether it's the wheel/driver at fault or simply Steam getting in the way.

Benjamin5678, 2021-12-14:
Hey its working now with the previously mentioned things I did and disabling steam input. Thank you all!

closed 2021-12-14 by Benjamin5678