scarburato/t150_driver #25: Getting Full Support for the T150 Pro version

Lazalatin, 2021-08-05:
Hi there,

I am new to the whole steering wheel business and just got myself a T150 Pro.
Is there any way I can help to expand support for it?

If there is, let me know :)

Regards,
Lazalatin

scarburato, 2021-09-22:
Just test it. The `hid` branch may support the shifter lever

vayerx, 2021-10-07:
Hi there,

There is a huge deadzone when using `hid` branch. `evdev-joystick` / `jscal` didn't help.

`master` is fine.

scarburato, 2021-10-08:
which firmware is installed?

vayerx, 2021-10-08:
`21`. I don't have Windows so I'm bound to this version until firmware update is available on Linux (_forever?..._).

isopix, 2021-10-09:
No, you are not.

You can download official Microsoft WinPE for free (which shudowns
after 48h of use) or any Rescue/Live system, based on it, and put it
on pendrive, like HirensBootCD:
https://www.hirensbootcd.org/

You can copy thrustmaster tools to the same pendrive and install them
in live and use, like in any other windows-based env.

Or you can complain to the rest of your live that you cannot do
something, because of 'Linux'...

Totten98, 2021-12-23:
Hello,
I've managed to solve the issue on #22 by using the `hid` branch.
Now FFB works ok.
Now I'm stuck with this "center deadzone" issue described on this issue. I noticed that changing the `ATTR{range}="1000"` to something smaller like 270 for example makes this center deadzone also smaller if this can help in some way :)
I'm really wishing to use my steering wheel but it feel closer but still far xD
I'm on Ubuntu 21.10 Kernel version 5.13.0-22.
I hope someone can fix this issue :P

Thanks, Totten98

Totten98, 2021-12-24:
Update: so I've been trying to make this thing work and even if on ETS2 native on Linux I get thiS center deadzone and it's aweful to play.
So out of ideas I tried to run ETS2 via Proton 6.3-8 and it works both the steering wheel and FFB flawlessly :D
I hope this can help someone.

Totten98

albfan, 2023-03-11:
Probably worth to mention hid branch is merged on master, and hid_tminit is merged into kernel since 5.13 (see #35)

Using hid_thrustmaster all looks fine (you can even feel the force feedback) but oversteer gets wheel and pedals wrong See (https://github.com/berarma/oversteer/issues/131)

Is there any progress to get t150 integrated into hid_thrustmaster? or force feedback will be always a different driver?

albfan, 2023-03-11:
I found there's a hid_tmff, for force feedback on thrustmaster, will this driver integrate with that at some point? is that the main goal?

albfan, 2023-03-12:
In this driver README it says:

> for example if you see input: Thrustmaster T150 steering wheel as /devices/pci0000:00/0000:00:14.0/usb1/1-1/input/input27
then the attributes will be located at sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/input/input27/device/.

but with hid_thrustmaster and hid_t150 working the path for this property files is different:

I get this path:

    /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:044F:B677.0291/input/input696

but 

    /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:044F:B677.0291/input/input696/device

points to

    /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:044F:B677.0291

and there, there are no such files

Searching with fd I found gain in

```
$ fd gain /sys/devices/
/sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/gain
```

looking at that path I found all the expected files:

```
$ ls -1 /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/
1-6:1.0
authorized
autocenter
avoid_reset_quirk
bcdDevice
bConfigurationValue
bDeviceClass
bDeviceProtocol
bDeviceSubClass
bmAttributes
bMaxPacketSize0
bMaxPower
bNumConfigurations
bNumInterfaces
busnum
configuration
descriptors
dev
devnum
devpath
driver
enable_autocenter
ep_00
firmware_node
firmware_version
gain
idProduct
idVendor
ltm_capable
manufacturer
maxchild
physical_location
port
power
product
quirks
range
removable
remove
rx_lanes
speed
subsystem
tx_lanes
uevent
urbnum
version
```

I'm unsure how this path is configured, but looks like it do not match the README expectation. I can try any tentative PR if needed

albfan, 2023-03-12:
After https://github.com/scarburato/t150_driver/commit/c217c777a4299c36ab582385434a53a5c8ffe9fe

axis names are provided by device itself, this conflicts with existing tools like oversteer:

https://github.com/berarma/oversteer/issues/131

not sure if any remapping option is available in this driver, or if there's a standard for wheel, accel, brake and clutch, any info to understand what code needs fixes?

albfan, 2023-03-19:
With latest version, hid report comes from hardware, and path for driver files (gain, autocenter, etc...) is in correct place. See d8ca058fa0a301aeb688ebeadbf6fca281ed3f9f

There're things to fix or add, but not specifically to T150 pro. T150 pro behaves the same as T150 so I suggest to close this issue and work on next improvements in driver in general, without specific device

closed 2023-03-20 by scarburato