scarburato/t150_driver #26: Error on install

BFCE, 2021-09-30:
`Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j12 KERNELRELEASE=5.11.22-tkg-bmq KDIR=/lib/modules/5.11.22-tkg-bmq/build all...(bad exit status: 2)
ERROR (dkms apport): binary package for t150: 0.7c not found
Error! Bad return status for module build on kernel: 5.11.22-tkg-bmq (x86_64)
Consult /var/lib/dkms/t150/0.7c/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j12 KERNELRELEASE=5.11.22-tkg-bmq KDIR=/lib/modules/5.11.22-tkg-bmq/build all...(bad exit status: 2)
ERROR (dkms apport): binary package for t150: 0.7c not found
Error! Bad return status for module build on kernel: 5.11.22-tkg-bmq (x86_64)
Consult /var/lib/dkms/t150/0.7c/build/make.log for more information.
==== INSTALLING UDEV RULES ====
'./files/etc/udev/rules.d/10-t150.rules' -> '/etc/udev/rules.d/10-t150.rules'
==== LOADING NEW MODULES ====
modprobe: FATAL: Module hid-t150 not found in directory /lib/modules/5.11.22-tkg-bmq
`
This is my output when trying to install the driver
The wheel works in game, but there is no force feedback, only a strong return to center

scarburato, 2021-10-01:
go in the repo directory and run

```
cd hid-t150
make all
```

Does it succeeds or it fails? (if it compiles you can manually load the driver with `insmod hid-t150.ko`)

Also, `5.11.22-tkg-bmq` are you using a custom kernel?

BFCE, 2021-10-01:
I am using a custom kernel. My normal kernel boot right now, otherwise I'd use it for test. 

`cd hid-t150
make all`
returns
`make -C /lib/modules/5.11.22-tkg-bmq/build M=/home/bfce/dev/t150_driver/hid-t150 modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.22-tkg-bmq'
make[2]: *** No rule to make target '/home/bfce/dev/t150_driver/hid-t150/hid-t150.o', needed by '/home/bfce/dev/t150_driver/hid-t150/hid-t150.mod'.  Stop.
make[1]: *** [Makefile:1820: /home/bfce/dev/t150_driver/hid-t150] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.11.22-tkg-bmq'
make: *** [Makefile:5: all] Error 2
`

scarburato, 2021-10-02:
mmmhhh can you try to compile hid-tminit? 
Clone it from [here](https://github.com/scarburato/hid-tminit) and run `make all` 

Maybe you don't have installed the appropriate `linux-headers` for your kernel

BFCE, 2021-10-02:
You were correct, I reinstalled my kernel and now the driver works perfect! thanks

closed 2021-10-02 by BFCE