scarburato/t150_driver #34: Dead zone on the Wheel

Mica-CRT, 2022-06-26:
Hi! i followed the steps to install correctly the t150 driver on my Manjaro Linux system, but from the games i tried and from the Manjaro's controller panel there is almost 45° dead-zone from each side starting from 0 and turning left/right, How could i fix this?

scarburato, 2022-06-30:
Try this: https://github.com/Kimplul/hid-tmff2#additional-tidbits the deadzone point

Put `b677` in place of `WHEEL_ID`

Mica-CRT, 2022-06-30:
thanks for the reply! I tried to put those parameters in my default rule and also tried to create another rule, reboot and reload modules but dead-zone still remain... thinking that the rules couldn't work i tried to modify other parameters but they work (such as steering range)

scarburato, 2022-07-01:
It's an joystick utility that it's typically found in the `joystick` package (on Arch it should be in `linuxconsole` instead) maybe you don't have it installed