scarburato/t150_driver #41: Implement switch for combined pedals

albfan, 2023-03-25:
Based on #40 probably combined pedals are set using command

0x42 05

Do we have any documentation for thrustmaster codes?

albfan, 2023-03-25:
I will implement a PR to test but confirm and see other addresses will make this easier

Kimplul, 2023-03-25:
> Do we have any documentation for thrustmaster codes?

Afraid not really, they seem to be exclusive to each wheel and not public information. Only known bits are what's been reverse engineered.

albfan, 2023-03-26:
Ok, have to do some test (like if setup is stored in device so windows driver reads it from config or similar) after that reserch (when I found how it works) we can think on how to implement it

albfan, 2023-03-30:
Seems this `0x42` codes are already documented but didn't change anything:

https://github.com/scarburato/t150_driver/blob/fdb83ef67aa544da92d81ab81a5fe45371168a16/hid-t150/settings.h#L15-L20

albfan, 2023-03-31:
So after debug with wireshark, hid report doesn't change (each pedal send same bytes), so this seems to be a software implementation, that redirects the report.

It can be reopen if someone miss this feature, but probably game or game manager (like steam) should deal with this

closed 2023-03-31 by albfan