scarburato/t150_driver #43: No force feedback as detected by SDL, but wheel autocenters when idle.

MomosuWarosu, 2023-04-25:
Quick system specs:
EndeavourOS "Cassini Nova"
Linux Kernel: 6.3.0 (linux-mainline, linux-mainline headers from AUR)
Shell: bash 5.1.16
CPU: Ryzen 5 5600X
GPU: Intel DG2 (Arc A770)
---
 
 So I have a Thrustmaster T150 Pro (with a TH8a shifter as well), and I installed the t150_driver-dkms-git package from the Arch User Repository. The install seemed to go fine, no errors, and all the package does is clone from this git repository and install the driver.

I'm trying to use this wheel with PCSX2, specifically pcsx2-git from the AUR which clones latest nightly (I'm doing it this way since running it from the AppImage doesn't seem too give it the correct permissions for devices for netplay.) I'm also on a newer kernel to fix HDMI audio for my GPU, as I am outputting to a TV like a real PS2.

After installing the driver, unplugging the wheel and plugging it back in again, it calibrates and autocenters like normal. When I tug on the wheel, it has an autocentering force. However when I start pcsx2-qt via console and check the log output it shows this when starting a game with force feedback commands:

```
[    1.4359] (SDLInputSource) Opened joystick 1 (instance id 1, player id 1): Thrustmaster T150RS
[    1.4359] (SDLInputSource) Rumble is supported on 'Thrustmaster T150RS' via haptic
...
[   40.8379] Opening USB...
[   40.8380] (SDLFFDevice) Friction effect is not supported on 'SDL-1'
[   40.8380] (SDLFFDevice) Autocenter effect is not supported on 'SDL-1'
```

The wheel goes limp whenever PCSX2 is open. No autocentering force, nothing. SDL-1 is the identifier assigned to the wheel by SDL2. The second I close PCSX2, the wheel autocenters again with force.

How can I figure out what's going wrong here? Why is PCSX2 not getting the force feedback capabilities of the wheel? Is it something to do with the udev rules?

albfan, 2023-04-25:
I would say all that is how other software use the driver. Probably open an issue on SDL or PCSX2 is the best idea.

Feedback from there could tell us if something is needed on driver.

Maybe you can do an usb capture to see what this apps send to device to configure it

MomosuWarosu, 2023-04-25:
Ohhh my god I just embarassed myself. Turns out I was selecting the wrong wheel device in PCSX2, meaning the game itself was not sending FF commands (which then, PCSX2 claims it doesn't support that effect.)
Tried Gran Turismo 4, which supports a greater range of wheels and it DOES work. Sorry about that!

closed 2023-04-25 by MomosuWarosu