scarburato/t150_driver #48: udev rules won't apply as they look in wrong path

dzosz, 2023-11-09:
I set up udev rules according to readme instructions. I'm getting following errors:
```
$ journalctl -u systemd-udevd
lis 09 17:47:56 tomek-laptop mtp-probe[14105]: checking bus 3, device 25: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4"
lis 09 17:47:56 tomek-laptop (udev-worker)[13808]: 3-4: /etc/udev/rules.d/t150-defaults.rules:1 Failed to write ATTR{/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4/range}, ignoring: No such file or directory
lis 09 17:47:56 tomek-laptop (udev-worker)[13808]: 3-4: /etc/udev/rules.d/t150-defaults.rules:1 Failed to write ATTR{/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4/gain}, ignoring: No such file or directory
```

My udev rule:

```SUBSYSTEM=="usb", ATTRS{idVendor}=="044f", ATTRS{idProduct}=="b677", ATTR{range}="540", ATTR{gain}="75"```

real path of device attrs is `/sys/devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:044F:B677.001D/range`

but udev rule looks here:
```
ls /sys/devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/
0003:044F:B677.001D  bAlternateSetting  bInterfaceNumber    bInterfaceSubClass  driver  ep_82          modalias           power      supports_autosuspend
authorized           bInterfaceClass    bInterfaceProtocol  bNumEndpoints       ep_01   firmware_node  physical_location  subsystem  uevent
```

albfan, 2023-11-10:
share dmesg when you connect the device, probably driver is not ruling the device

dzosz, 2023-11-11:
> share dmesg when you connect the device, probably driver is not ruling the device

yeah something strange happens with xhci_hcd module

```
[   39.989922] hid_t150: loading out-of-tree module taints kernel.
[   68.521336] usb 3-4: new full-speed USB device number 10 using xhci_hcd
[   68.649648] usb 3-4: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[   68.649653] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   68.649654] usb 3-4: Product: Thrustmaster FFB Wheel
[   68.649655] usb 3-4: Manufacturer: Thrustmaster
[   69.190019] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:044F:B65D.000C/input/input29
[   69.190163] hid-thrustmaster 0003:044F:B65D.000C: input,hidraw8: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:00:14.0-4/input0
[   69.208185] hid-thrustmaster 0003:044F:B65D.000C: Wheel with model id 0x306 is a Thrustmaster T150RS
[   69.211260] hid-thrustmaster 0003:044F:B65D.000C: Success?! The wheel should have been initialized!
[   69.211331] usb 3-4: USB disconnect, device number 10
[   69.500362] usb 3-4: new full-speed USB device number 11 using xhci_hcd
[   69.628374] usb 3-4: New USB device found, idVendor=044f, idProduct=b677, bcdDevice= 1.00
[   69.628379] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   69.628380] usb 3-4: Product: Thrustmaster T150RS
[   69.628381] usb 3-4: Manufacturer: Thrustmaster
[   69.630832] input: Thrustmaster Thrustmaster T150RS as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:044F:B677.000D/input/input30
[   69.631074] hid-t150 0003:044F:B677.000D: input,hidraw8: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T150RS] on usb-0000:00:14.0-4/input0
[   69.643799] hid-t150 0003:044F:B677.000D: Setup completed! Firmware version is 35
[   74.820511] warning: `ThreadPoolForeg' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
```

albfan, 2023-11-12:
Your dmesg seems ok.

So I dig into this a bit:

```
$ fd -p .*B677.*/range /sys
/sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:044F:B677.0022/range
```

Using that let's check the udevadm info:

```
$ sudo udevadm info --attribute-walk /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:044F:B677.0022

Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:044F:B677.0022':
    KERNEL=="0003:044F:B677.0022"
    SUBSYSTEM=="hid"
    DRIVER=="hid-t150"
    ATTR{autocenter}=="50"
    ATTR{country}=="00"
    ATTR{enable_autocenter}=="n"
    ATTR{firmware_version}=="35"
    ATTR{gain}=="80"
    ATTR{power/control}=="auto"
    ATTR{power/runtime_active_time}=="0"
    ATTR{power/runtime_status}=="unsupported"
    ATTR{power/runtime_suspended_time}=="0"
    ATTR{range}=="1080"
...
```

so 

```
SUBSYSTEM=="hid", ATTR{range}="540", ATTR{gain}="75"
```

should work. Are you able to provide a PR to fix this documentation?

albfan, 2023-11-13:
Fixed on 9ae60e7

closed 2023-11-13 by albfan