scarburato/t150_driver #53: T80 support

TorokLev, 2024-03-30:
Hi,
I know that this is not supposed to work with T80 but I'd like to ask for some help how build a support for T80.
Can you help me in this?
Thank you,
Lev

albfan, 2024-03-30:
t80 is basically t150 without force feedback.

I would start with oversteer. Seems they add support since version 0.8.0.

https://github.com/berarma/oversteer/releases/tag/0.8.0

Remember to use PS3 mode (like this driver)

https://github.com/berarma/oversteer/issues/142

seing that, I don't think you need a driver for it, it is just buttons and analog input

closed 2024-03-30 by albfan

TorokLev, 2024-03-30:
Hi,
Thanks for the suggestion.

Unfortunately, this device works in PS4 PS5 mode and no way to switch to PS3.

Oversteer can see only 2 buttons. evtest can detect more but not the wheel and pedals.

Any other suggestion?
Lev

albfan, 2024-03-30:
Low level work. USB captures, protocol analysis

Maybe tmff2 supports PS4 mode (no plans for that here)
 https://github.com/Kimplul/hid-tmff2/issues/86