scarburato/t150_driver #58: Deprecation warnings during kernel update

dusanpol, 2024-09-07:
Hi, I noticed a warning when I updated my kernel. Will this be a problem in future?

```
 * dkms: running auto installation service for kernel 6.10.8-3-liquorix-amd64
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/t150/0.8a/source/dkms.conf)
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/t150/0.8a/source/dkms.conf)
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/t150/0.8a/source/dkms.conf)
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/t150/0.8a/source/dkms.conf)
```

albfan, 2024-09-08:
Can you give us detrails on your environment.

I found related info on https://github.com/openzfs/zfs/issues/849

Which looks really specific to ZFS, so removing It shouldn't be a problem.

katnatek, 2025-02-19:
> Hi, I noticed a warning when I updated my kernel. Will this be a problem in future?
>
> ```
>  * dkms: running auto installation service for kernel 6.10.8-3-liquorix-amd64
> Deprecated feature: [...]
I have similar issue
I change REMAKE_INITRD=yes to REMAKE_INITRD=no in dkms.conf

albfan, 2025-02-21:
dkms sources are on: https://github.com/dell/dkms

The only argument is that 4 years ago, someone found this is not in use anymore

https://github.com/dell/dkms/commit/961f9ceac4cf6772cb4c52bf4836a0f6c9818239

So let's remove that obsolete config and have a cleaner build.

fixed on 8266ddd

closed 2025-02-21 by albfan