scarburato/t150_driver #61: No force feedback with Linux Kernel x86_64 6.11.10

bokonon-foma, 2025-01-11:
Force Feeback stopped working for T150 in Batocera when kernel updated to 6.11.10.  Worked fine with kernel 6.9.10.  After doing some trouble shooting with lsmod and dmesg, Batocera developers speculate that perhaps t150_driver uses something from another driver which changed.

scarburato, 2025-01-15:
Hello, I'm using `6.12.9-arch1-1` and it works just fine.

Is DKMS working properly? What's the output of `dkms status`?

If you run `dmesg -w` and connect the wheel, what does it print? If you run `evtest` can you see the steering wheel? Does the spring effect disappear as soon as you open the device in evtest? If you run `fftest` and/or `ffmvforce` --- you'll have to provide the device path you saw in evtest as their first argument, --- does the ffb work?

scarburato, 2025-01-15:
And, most importantly, did you put your wheel in PS3 mode, right?

closed 2025-01-24 by scarburato