scarburato/t150_driver #62: Module.symvers file not found

katnatek, 2025-02-21:
I'm trying to import this for mageia but when I install the package I get the message Module.symvers file not found
The process create an empty file in the folder build inside the module sources

How I fix this?, Is something to be worried?

albfan, 2025-02-22:
mageia is a linux distro: https://www.mageia.org/es/

Module.symvers is a file created on kernel

https://www.quora.com/What-is-the-purpose-of-Module-symvers-in-Linux

```
In a nutshell Module.symvers contains a list of all exported symbols from a kernel build.

During a kernel build the symvers or symbol versions file will be generated. If you pop it open, the general syntax of the file is :

<CRC>			<Symbol>				<module> 
0x2d036834		scsi_remove_host		drivers/scsi/scsi_mod 
If “CONFIG_MODVERSIONS” is disabled then the CRC would contain an entry 0x00000000.

Modules.symvers serves two purposes:

It lists all exported symbols from vmlinux and all modules.
It lists the CRC if CONFIG_MODVERSIONS is enabled.
When building an external module, the build system needs access to the symbols from the kernel to check if all external symbols are defined. This is done in the MODPOST step. modpost obtains the symbols by reading Module.symvers from the kernel source tree. If a Module.symvers file is present in the directory where the external module is being built, this file will be read too. During the MODPOST step, a new Module.symvers file will be written containing all exported symbols that were not defined in the kernel.

For further information : [/kbuild/modules.txt](https://www.kernel.org/doc/Documentation/kbuild/modules.txt) , in the linux kernel documentation.
```

So you probably want to contact package maintainer on mageia to resolve your question.

You can edit your description with exact output error you see (for clarification), but until mageia maintainer don't expose something we need to fix here, closing as out of scope

closed 2025-02-22 by albfan

katnatek, 2025-02-23:
Not exist maintainer yet
This is the output of the installing the package I create (I use --verbose to get more information)

Preparing...                     ##################################################################################################
      1/1: t150_driver           ##################################################################################################

Creating symlink /var/lib/dkms/t150_driver/0.8a-1.20250203git580b79b.mga9/source ->
                 /usr/src/t150_driver-0.8a-1.20250203git580b79b.mga9

DKMS: add Completed.

Preparing kernel 6.6.79-1.mga9 for module build:
(This is not compiling a kernel, just preparing kernel symbols)
Storing current .config to be restored when complete
Running Generic preparation routine
make mrproper
make[1]: No se hace nada para 'mrproper'.

using /proc/config.gz
make KERNELRELEASE=6.6.79-1.mga9 oldconfig
make[1]: No se hace nada para 'oldconfig'.

make -j4 KERNELRELEASE=6.6.79-1.mga9 prepare scripts
  SYNC    include/config/auto.conf
make[2]: No se hace nada para 'syncconfig'.
make[1]: No se hace nada para 'prepare'.
make[1]: No se hace nada para 'scripts'.

Building module:
make KDIR=/lib/modules/6.6.79-1.mga9/build clean
make -C ./hid-t150 clean
make[1]: Entering directory '/var/lib/dkms/t150_driver/0.8a-1.20250203git580b79b.mga9/build/hid-t150'
make -C /lib/modules/6.6.79-1.mga9/build M=/var/lib/dkms/t150_driver/0.8a-1.20250203git580b79b.mga9/build/hid-t150 clean
make[2]: Entering directory '/usr/src/kernel-linus-devel-6.6.79-1.mga9'
make[2]: Leaving directory '/usr/src/kernel-linus-devel-6.6.79-1.mga9'
make[1]: Leaving directory '/var/lib/dkms/t150_driver/0.8a-1.20250203git580b79b.mga9/build/hid-t150'
rm -rf ./build/*

/sbin/dkms: line 1416: build/Module.symvers: No such file or directory
make -j4 KERNELRELEASE=6.6.79-1.mga9 KDIR=/lib/modules/6.6.79-1.mga9/build all >> /var/lib/dkms/t150_driver/0.8a-1.20250203git580b79b.mga9/build/make.log 2>&1

make KDIR=/lib/modules/6.6.79-1.mga9/build clean
make -C ./hid-t150 clean
make[1]: Entering directory '/var/lib/dkms/t150_driver/0.8a-1.20250203git580b79b.mga9/build/hid-t150'
make -C /lib/modules/6.6.79-1.mga9/build M=/var/lib/dkms/t150_driver/0.8a-1.20250203git580b79b.mga9/build/hid-t150 clean
make[2]: Entering directory '/usr/src/kernel-linus-devel-6.6.79-1.mga9'
  CLEAN   /var/lib/dkms/t150_driver/0.8a-1.20250203git580b79b.mga9/build/hid-t150/Module.symvers
make[2]: Leaving directory '/usr/src/kernel-linus-devel-6.6.79-1.mga9'
make[1]: Leaving directory '/var/lib/dkms/t150_driver/0.8a-1.20250203git580b79b.mga9/build/hid-t150'
rm -rf ./build/*

make mrproper
make[1]: No se hace nada para 'mrproper'.

DKMS: build Completed.

hid-t150.ko.xz:
 - Installation
   - Installing to /lib/modules/6.6.79-1.mga9/dkms/drivers/input/joystick/

depmod -au 6.6.79-1.mga9 -F /boot/System.map-6.6.79-1.mga9
depmod: WARNING: Ignored deprecated option -u

katnatek, 2025-02-23:
In mageia think the bug is in your side
https://bugs.mageia.org/show_bug.cgi?id=33899#c23
https://stackoverflow.com/a/29606599

katnatek, 2025-02-24:
> Mageia issue tracker says your machinery to build package is complaining about Module.symvers not present:
> 
> > /sbin/dkms: line 1416: build/Module.symvers: No such file or directory
> 
> My 1416 line on dkms is empty
> 
> ```
Did you dkms version is >= 3 ?, In mageia still use 2.0.19, if that is the cause, ill report

> On a personal view, a package that includes oversteer, and several hid drivers, is the wrong way to maintain it for a distro, just saying.
> 
oversteer and each module is handled in separated packages, but as are of aprentship  are included in same bug

> See we don't surrender on downstream packages, even when they try to put the blame on us.
> 
> Patches welcome, but first fix on your side, provide a patch and let's discuss here upstream.
> 
> ```
> Source0:	%{url}/archive/%{commit0}/%{name}-%{shortcommit0}.tar.gz
> #Patch0:    Module.Symvers.patch
> ```
> 
> ;)

The commented patch is an attempt to see if add KBUILD_EXTRA_SYMBOLS helps, but I couldn't get rid of the message and  even broke the module build :(

albfan, 2025-02-25:
Cool, Let's keep this closed. We can open a new one about improvements on buildsystem config it those clean really break something

katnatek, 2025-02-25:
> Cool, Let's keep this closed. We can open a new one about improvements on buildsystem config it those clean really break something

I'm quite sure the file must not be deleted, the process store the current .config, generates a new one and should restore the original .config when uninstall the package/module

albfan, 2025-02-26:
I see uninstall step on other drivers, but here we don't have it. That might be a valid new issue to work on

katnatek, 2025-02-26:
I could not reproduce again the .config issue (I make sure of remove the patch) :S , if happen again I'll report