scarburato/t150_driver #63: spring, damper, friction effect

albfan, 2025-02-22:


Kimplul, 2025-02-22:
If I understand your question correctly, yeah, the control panel values (Oversteer/Thrustmaster control panel) are not directly sent to the wheel, instead they get applied to a force effect when a game tries to send one out. The wheel itself doesn't apply any corrections, as far as I'm aware.

albfan, 2025-02-23:
Using a stock photo for reference (my panel is in spanish)

[image]

sliders for constant, periodic, spring and damper forces only works when effect is active.

each time you focus in/out from control panel a `0x4205` is sent.

if you start an effect like "Blown tire" effect this data is sent:

41 01 41 01
41 00 41 01 

At this point sliders send info if you move them.

When you focus out, it stop too the effects:

42 05
41 01 00 01
41 00 00 01 

then sliders do not send anything anymore

Different effects attend to different sliders, I couldn't find any effect attending to spring force (which is an effect for a supposed spring on wheel, so might make sense for that list of effects)

fftest might help here, but I guess we can found by discard

When "Blown tire" is enabled, moving constant force driver sent (last hex byte):

03 1e 03 ff (0%)
03 1e 03 c0 (100%)

255 - 192 =63 possible values. Looks kind strange, but, 50% value is e0 (224) 255 -224 = 31, middle of those values

For periodic forces it is (last hex byte):

02 c0 01 d9 00 00 aa 02 00 (0%) 0
02 c0 01 d9 00 00 aa 02 2c (50%) 44
02 c0 01 d9 00 00 aa 02 58 (100%) 88

Punch effect allows to expose damper force (repeated on 4 and 5 hex byte):

05 ae 02 00 00 00 00 00 00 64 64 (0%) 0
05 ae 02 32 32 00 00 00 00 64 64 (50%) 50
05 ae 02 64 64 00 00 00 00 64 64 (100%) 100

second hex byte can change ae, e6, not sure how that affects

At least to implement this controls it might be enough. Let's add a fftest dump to compare what linux driver launch

wheelcheck shows too valuable info, and shows controls for spring force

[image]

With all that info, there's a start setup every time we start an effect, where all these current values are sent:

Effect "Blow tire" send  (all forces 100%):

41 00 00 01

02 48 03 d9 00 00 aa 02 58 (periodic 100%)
04 3a 03 0a 00 00 22 02
01 00 20 40 e2 04 00 00 00 3a 03 48 03 00 00

41 01 00 01

02 64 03 00 00 7f 00 00 7f
03 56 03 c0  (constant force 100%)
01 01 00 40 84 03 00 00 00 56 03 64 03 00 00

Not always it send 3 messages, but as some of them are recognizable, probaby we can identify what the other messages are related with.

On wheelcheck I see several concepts present on tmff2 driver (coefficient, saturation) so maybe we can share same implementation (or merge drivers at some point) even merge upstream @Kimplul?

albfan, 2025-02-23:
https://github.com/flosse/linuxconsole/blob/master/utils/fftest.c#L277

Just enabling spring effect you can feel how a free wheel start to apply some force. We can replicate that effect and see hid messages sent from windows and fftest till this gets implemented

Kimplul, 2025-02-23:
> On wheelcheck I see several concepts present on tmff2 driver (coefficient, saturation) so maybe we can share same implementation (or merge drivers at some point) even merge upstream @Kimplul?

Sure, I'm all for it. There was a bit of discussion about merging T150 functionality into `hid-tmff2` in https://github.com/Kimplul/hid-tmff2/issues/78, but I don't know what the status on that is.

albfan, 2025-05-15:
Exciting, would be awesome to reach upstream and merge all this work.