scarburato/t150_driver #65: BeamNG (linux native) and other linux native games either don't pick up the wheel, or don't work with force feedback

cydiralis, 2025-05-15:
As the title states, some games that are linux native dont pick up the wheel / don't work with force feedback:

e.g

dirt 4: doesnt pick up the wheel
beamng linux native: no force feedback, ffb works on the windows version fine, but i have flickering issues on the windows version so i cant use it

scarburato, 2025-05-15:
works on my machine, does the problem persist with the Steam client closed?.

On an unrelated note, BeamNG's flickering is an issue with their Vulkan backend, limiting the game FPS to like 80 should fix it for now

cydiralis, 2025-05-15:
tried a relaunch and now im back at square one with ffb not working at all

scarburato, 2025-05-15:
try `fftest` and `ffmvforce`. Also in the game's setting try reducing the ffb update rate

cydiralis, 2025-05-15:
both fftest and ffmvforce work, nothing i do in beam works however

cydiralis, 2025-05-16:
hello, any updates on this thanks :)

closed 2025-05-16 by scarburato