scarburato/t150_driver #66: Upstreaming Questions

pastaq, 2025-05-18:
I'm currently working through the codebase for the upstreaming effort I mentioned in #63 and had a few questions. I'll probably have more so I'll use this as a single thread until I'm able to upstream.
 
Can I get some background for this comment in the ffb code:
https://github.com/scarburato/t150_driver/blob/f7ecb30c65ee5f7870e921bc0a2354df8e1e8100/hid-t150/forcefeedback.c#L123
How/why these are not correct, if we know the correct values, or what we need to get them would be helpful if possible.

Similarly, what still needs TODO here?
https://github.com/scarburato/t150_driver/blob/f7ecb30c65ee5f7870e921bc0a2354df8e1e8100/hid-t150/input.c#L79

For the urb array, is there a reason this was used? 
https://github.com/scarburato/t150_driver/blob/f7ecb30c65ee5f7870e921bc0a2354df8e1e8100/hid-t150/hid-t150.h#L30

Per the [docs](https://www.kernel.org/doc/html/v5.10/driver-api/usb/usb.html), this doesn't appear strictly necessary, but I expect I'm probably misunderstanding the meaning of: 

```
"URBs may be submitted to endpoints before previous ones complete, to minimize the impact of interrupt latencies and system overhead on data throughput. "
```

I'm wondering if issues were experienced when originally writing this driver that led to the use of an array instead of reusing a single URB? I'll end up trying both ways, but some insight would be great. The docs say this was added in kernel 2.6.

More to follow, thanks.

scarburato, 2025-05-22:
Sorry, I don't really remember the details, maybe I didn't bother to fully reverse-engineer the behavior of the Windows driver.

I might've put that _TODO_ because I didn't bother to return a proper error code.

Regarding the URBs, I think they're necessary. You cannot modify the URB's data while an operation is pending:

> With few exceptions, USB device drivers should never access URB fields provided by usbcore or the HCD until its complete() is called.

you can submit other instance of a URB data-structure, not the same one. The pointer to the URB is used directly the USB subsystem to store information relative to the request

https://elixir.bootlin.com/linux/v6.14.7/source/drivers/usb/core/urb.c#L368

If you want to submit n different requests concurrently, you need n instances of the URB structure.

albfan, 2025-10-20:
Could make sense to quick review those question and cleanup comments if don't recall anything, I will do and provide a related PR